kerberos-io/machinery #218: Error Configuring security camera

djmgrant, 2020-11-15:
Kerberos.io not displaying camera output.  

Kerberos Log : 
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 20
- streams.Mjpg.password = 
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- streams.Mjpg.username = 
- timezone = Europe-London
15/11/2020 11:11:15.228 INFO  [trivial] Logging is set to info
15/11/2020 11:11:15.229 INFO  [trivial] Capture: Start capture device: RaspiCamera
15/11/2020 11:11:15.229 ERROR [trivial] Kerberos : Factory : Could not create an instance of message: 

Steps to reproduce the behavior:
1. Running Balena v2.61.3 on Raspberry Pi 4 - Camera is Longrunner Raspi Pi4 Model B+ Camera module with switching.  
    Installed using Balena etcher 
2. On Balena I've set BALENA_HOST_CONFIG_start_x = 1 plus also set GPU Memory settings as suggested in other posts.  
3.  On the Raspi command line i can see the camera is now working : 

vcgencmd get_camera 
supported=1 detected=1

3. In Kerberos.io System is running and ive set the config as follows.  
[image]

Here is the content of my config.  (this has not been edited) 
<?xml version="1.0"?>
<kerberos>

    <instance>
        <name type="text">hall</name>
        <logging type="bool">false</logging>
        <timezone type="timezone">Europe-London</timezone>
        <capture file="capture.xml">RaspiCamera</capture>
        <stream file="stream.xml">Mjpg</stream>
        <condition file="condition.xml" type="multiple">Enabled</condition>
        <algorithm file="algorithm.xml">DifferentialCollins</algorithm>
        <expositor file="expositor.xml">Hull</expositor>
        <heuristic file="heuristic.xml">Sequence</heuristic>
        <io file="io.xml" type="multiple">Video</io>
        <cloud file="cloud.xml">S3</cloud>
    </instance>

</kerberos>

Here are the capture settings.  
<?xml version="1.0"?>
<captures>

    <IPCamera>
        <url type="text">rtsp://admin:888888@192.168.0.13/tcp/av0_1</url>
        <frameWidth type="number">640</frameWidth>
        <frameHeight type="number">480</frameHeight>
        <delay type="number">500</delay>
        <angle type="number">0</angle>
    </IPCamera>

    <USBCamera>
        <frameWidth type="number">640</frameWidth>
        <frameHeight type="number">480</frameHeight>
        <deviceNumber type="number">0</deviceNumber>
        <fourcc type="text">MJPG</fourcc>
        <delay type="number">500</delay>
        <angle type="number">0</angle>
    </USBCamera>

    <RaspiCamera>
        <frameWidth type="number">640</frameWidth>
        <frameHeight type="number">480</frameHeight>
        <delay type="number">500</delay>
        <angle type="number">0</angle>
        <framerate type="number">20</framerate>
        <sharpness type="number">0</sharpness>
        <saturation type="number">0</saturation>
        <contrast type="number">0</contrast>
        <brightness type="number">50</brightness>
    </RaspiCamera>

    <VideoCapture>
        <frameWidth type="number">640</frameWidth>
        <frameHeight type="number">480</frameHeight>
        <path type="text">0</path>
        <delay type="number">500</delay>
        <angle type="number">0</angle>
    </VideoCapture>

</captures>

djmgrant, 2020-11-17:
So got this working by selecting IP camera.   Not sure why this is specific to Pi4 ,  Pi3 RPI camera works.

cedricve, 2020-11-17:
hey @djmgrant sorry for the late reply, you made it working by enabling IP Camera setting?

djmgrant, 2020-11-18:
Yes thats correct.  Quite weird really.  My friend bought the exact same camera and on raspberry pi 3 his camera worked as an Rpi camera.  Not a big deal but took me quite a while to work this out,  swapping ribbons and power sources etc etc.