kerberos-io/machinery #219: Machinery does not run in arm/v7 dockerimage

GittyDan, 2020-12-21:
I tried the kerberos/kerberos:latest dockerimages on two architectures. both times i started a blank container.

Docker for Windows: **linux/amd64**
[image]

Qnap TS-431P2 NAS: **linux/arm/v7**
[image]

Here the console output of the container:
```
docker run -p 801:80 -p 8889:8889 kerberos/kerberos:latest
Config files are available.
Config files are available.
2020-12-21 22:26:25,561 CRIT Set uid to user 0
2020-12-21 22:26:25,590 INFO RPC interface 'supervisor' initialized
2020-12-21 22:26:25,590 CRIT Server 'unix_http_server' running without any HTTP authentication checking
2020-12-21 22:26:25,590 INFO supervisord started with pid 27
2020-12-21 22:26:26,594 INFO spawned: 'nginx' with pid 30
2020-12-21 22:26:26,597 INFO spawned: 'machinery' with pid 31
2020-12-21 22:26:26,769 INFO exited: machinery (exit status 255; not expected)
2020-12-21 22:26:27,771 INFO success: nginx entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
2020-12-21 22:26:27,773 INFO spawned: 'machinery' with pid 38
2020-12-21 22:26:27,948 INFO exited: machinery (exit status 255; not expected)
2020-12-21 22:26:29,953 INFO spawned: 'machinery' with pid 41
2020-12-21 22:26:30,122 INFO exited: machinery (exit status 255; not expected)
2020-12-21 22:26:33,128 INFO spawned: 'machinery' with pid 44
```
The machinery is exiting all the time.

lexustec, 2021-02-02:
Hey @GittyDan ,

running in the same issue. Did You find a sollution. I looked into the docker department and didnt find the issue issued there. I assume you solved it.

Can You give a hint if so ?

Many greetings

Technaur, 2021-02-12:
I am running into the same issue on Raspberry Pi 4. Any solution?
[image]

cedricve, 2021-02-12:
Thanks for reporting all, could one of you share the machinery logs?

Technaur, 2021-02-12:
I am not sure if these are the Machinery logs, but here are the logs from the System view, hope it helps.

```
- expositors.Rectangle.region.x1 = 0
- expositors.Rectangle.region.x2 = 800
- expositors.Rectangle.region.y1 = 0
- expositors.Rectangle.region.y2 = 600
- heuristic = Sequence
- heuristics.Counter.appearance = 3
- heuristics.Counter.markers = 100,100|100,200|200,100|200,200
- heuristics.Counter.maxDistance = 140
- heuristics.Counter.minArea = 200
- heuristics.Counter.minimumChanges = 20
- heuristics.Counter.noMotionDelayTime = 100
- heuristics.Counter.onlyTrueWhenCounted = true
- heuristics.Sequence.minimumChanges = 20
- heuristics.Sequence.minimumDuration = 2
- heuristics.Sequence.noMotionDelayTime = 1000
- io = Video
- ios.Disk.directory = /etc/opt/kerberosio/capture/
- ios.Disk.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token.jpg
- ios.Disk.markWithTimestamp = false
- ios.Disk.privacy = false
- ios.Disk.throttler = 0
- ios.Disk.timestampColor = white
- ios.GPIO.periodTime = 100000
- ios.GPIO.periods = 1
- ios.GPIO.pin = 17
- ios.GPIO.throttler = 0
- ios.MQTT.clientId = 
- ios.MQTT.password = 
- ios.MQTT.port = 1883
- ios.MQTT.secure = false
- ios.MQTT.server = 127.0.0.1
- ios.MQTT.throttler = 0
- ios.MQTT.topic = kios/mqtt
- ios.MQTT.username = 
- ios.MQTT.verifycn = false
- ios.Pushbullet.throttler = 10
- ios.Pushbullet.token = o.mC5LPVCvPCphtSsEgWZQpFM86w9ciWQ3
- ios.Pushbullet.url = https://api.pushbullet.com
- ios.Script.path = /etc/opt/kerberosio/scripts/run.sh
- ios.Script.throttler = 0
- ios.TCPSocket.message = motion-detected
- ios.TCPSocket.port = 1337
- ios.TCPSocket.server = 127.0.0.1
- ios.TCPSocket.throttler = 0
- ios.Video.codec = h264
- ios.Video.directory = /etc/opt/kerberosio/capture/
- ios.Video.enableHardwareEncoding = true
- ios.Video.extension = mp4
- ios.Video.fileFormat = timestamp_microseconds_instanceName_regionCoordinates_numberOfChanges_token
- ios.Video.fps = 30
- ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 30
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.throttler = 0
- ios.Video.timestampColor = white
- ios.Webhook.throttler = 0
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = false
- name = frontdoor
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 15
- streams.Mjpg.password = 
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
[... 109 lines cut ...]
- ios.Video.hardwareDirectory = /etc/opt/kerberosio/h264/
- ios.Video.markWithTimestamp = false
- ios.Video.maxDuration = 30
- ios.Video.privacy = false
- ios.Video.recordAfter = 5
- ios.Video.throttler = 0
- ios.Video.timestampColor = white
- ios.Webhook.throttler = 0
- ios.Webhook.url = http://localhost/api/v1/webhook
- logging = false
- name = frontdoor
- stream = Mjpg
- streams.Mjpg.enabled = true
- streams.Mjpg.fps = 15
- streams.Mjpg.password = 
- streams.Mjpg.quality = 75
- streams.Mjpg.streamPort = 8889
- streams.Mjpg.username = 
- timezone = Europe-Brussels
12/02/2021 07:42:42.041 INFO  [trivial] Logging is set to info
12/02/2021 07:42:42.041 INFO  [trivial] Capture: Start capture device: RaspiCamera
12/02/2021 07:42:42.042 INFO  [trivial] Capture: Trying to open Raspberry Pi camera module.
12/02/2021 07:42:42.042 INFO  [trivial] Capture: (Warning) If you see a OMX_GetHandle error, this means that you don't have a working RPi camera module attached.
12/02/2021 07:42:42.042 INFO  [trivial] Capture: (Warning) You can change the capture device with the configuration files.
```

cedricve, 2021-02-12:
Are you connecting a Raspberry Pi camera?

Technaur, 2021-02-12:
No, I want to use an external IP camera with an RTSP stream

cedricve, 2021-02-12:
Then you should change it to IPCamera. Currently it is set to Raspberry Pi camera.

    - capture = RaspiCamera

Technaur, 2021-02-12:
Ah okay. I see. In my case, I wanted to use an IP camera, but because it was still set to RaspiCamera (That was not connected) it failed.

How I fixed it:
- Set up the IP camera first and make sure the config for the IP camera is saved correctly.
- Restart the docker container
- It worked, I can see my live stream from the IP Camera. (I did not realize that a broken cam config would cause Machinery to not run, must have missed it)

Thanks for the help @cedricve. Much appreciated. :)

cedricve, 2021-02-12:
No worries we are adding an autodetect or waning in the next major release 3.0. So it is more obvious.
https://github.com/kerberos-io/opensource