kerberos-io/machinery #223: Browser notification can't be disabled!

jackstuard, 2021-02-19:
**Describe the bug**
Notifications are always enabled, and the speech api can't even be disabled by chrome "mute tabs"
Please allow it to be disabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to main screen and wait.

**Expected behavior**
Disable toggle really disable that speech synthesis notifications

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Operating System and Device (please complete the following information):**
 - Host system: Linux VM, Raspberry Pi Version: [0, 1, 2, 3]
 - OS: KiOS, Raspbian, Docker, Ubuntu, CentOS, ..
 - Installation Method: KiOS installer, Etcher, Generic Installation, Docker

**Additional context**
Add any other context about the problem here.

cedricve, 2021-02-19:
@jackstuard you have to go to the livestream page, and press the voice toggle at the right bottom. We will make that clear in the next version of Kerberos Cloud.

jackstuard, 2021-02-19:
That works! Thank you!
Please maybe tie with this option, it will be more clear:
[image]

cedricve, 2021-02-20:
Good point thank you @jackstuard

closed 2021-02-20 by cedricve